beegee-tokyo/SX126x-Arduino #110: IRQ_HEADER_ERROR should raise the RxError event in RadioBgIrqProcess (Radio.cpp)

JeromeBriot, 2023-11-21:
Hello,

The block of code below in RadioBgIrqProcess (Radio.cpp):

```cpp
if ((irqRegs & IRQ_HEADER_ERROR) == IRQ_HEADER_ERROR)
{
	LOG_LIB("RADIO", "RadioIrqProcess => IRQ_HEADER_ERROR");

	TimerStop(&RxTimeoutTimer);
	if (RxContinuous == false)
	{
		//!< Update operating mode state to a value lower than \ref MODE_STDBY_XOSC
		SX126xSetOperatingMode(MODE_STDBY_RC);
	}
	if ((RadioEvents != NULL) && (RadioEvents->RxTimeout != NULL))
	{
		RadioEvents->RxTimeout();
	}
}
```

Should be:

```cpp
if ((irqRegs & IRQ_HEADER_ERROR) == IRQ_HEADER_ERROR)
{
	LOG_LIB("RADIO", "RadioIrqProcess => IRQ_HEADER_ERROR");

	TimerStop(&RxTimeoutTimer);
	if (RxContinuous == false)
	{
		//!< Update operating mode state to a value lower than \ref MODE_STDBY_XOSC
		SX126xSetOperatingMode(MODE_STDBY_RC);
	}
	if ((RadioEvents != NULL) && (RadioEvents->RxTimeout != NULL))
	{
		RadioEvents->RxError();
	}
}
```

What do you think?

beegee-tokyo, 2023-11-22:
Yes, that make sense. With a small change:
```cpp
if ((RadioEvents != NULL) && (RadioEvents->RxTimeout != NULL))
```
should be 
```cpp
if ((RadioEvents != NULL) && (RadioEvents->RxError!= NULL))
```
Otherwise it would lead to a crash if RxError function is not set.

beegee-tokyo, 2023-11-22:
Implemented in V2.0.22

closed 2023-11-22 by beegee-tokyo