beegee-tokyo/SX126x-Arduino #116: why is synchronization not used in the library?

nic2019, 2024-01-29:
Hi,  
Question 1:
I am programming under RAK4631 (nRF52832) in Arduino (RAKwireless-Arduino-BSP-Index (With block RAK4631  v1.3.3)   FREERTOS ). Why is there no synchronization blocks (global variables) between timers and tasks in your library code (OnMacStateCheckTimerEvent, OnRxWindow1TimerEvent, OnRxWindow2TimerEvent, OnTxDelayedTimerEvent, .., AND  start_lora_task,  and main loop (sending msg) )?  
 In the C++ specification, this is a data race - undefined behavior.

beegee-tokyo, 2024-01-29:
Not sure I understand what you mean. There is no direct communication between the application layer and the LoRaWAN stack.

Initializing LoRaWAN or sending a LoRaWAN packet is initiated by calling the LoRaMacHelper functions.
LoRaWAN events are handled by the lora task inside the library.
Every result of LoRaWAN events is sent back to the application layer using the callbacks defined with 
```cpp
 lmh_callback_t lora_callbacks = {get_lora_batt, BoardGetUniqueId, BoardGetRandomSeed, lpwan_rx_handler,
										lpwan_joined_handler, lpwan_class_confirm_handler, lpwan_join_fail_handler,
										lpwan_unconfirm_tx_finished, lpwan_confirm_tx_finished};
```
I don't see a need for using global variables. The application layer should not interfere with the LoRaMac stack directly.

nic2019, 2024-01-29:
(snippet LoRaMac.cpp):
...
TimerInit(&MacStateCheckTimer, OnMacStateCheckTimerEvent);
TimerInit(&TxDelayedTimer, OnTxDelayedTimerEvent);
TimerInit(&RxWindowTimer1, OnRxWindow1TimerEvent);
TimerInit(&RxWindowTimer2, OnRxWindow2TimerEvent);
TimerInit(&AckTimeoutTimer, OnAckTimeoutTimerEvent);
// Initialize Radio driver
RadioEvents.TxDone = OnRadioTxDone;
RadioEvents.RxDone = OnRadioRxDone;
RadioEvents.RxError = OnRadioRxError;
RadioEvents.TxTimeout = OnRadioTxTimeout;
RadioEvents.RxTimeout = OnRadioRxTimeout;
Radio.Init(&RadioEvents);
...
-------------
The start_lora_task function creates the "LORA" task (381 in the boards\mcu\board.cpp file) in which the Radio.BgIrqProcess() function is called (essentially a call to RadioBgIrqProcess).

  Functions are called from the RadioBgIrqProcess : RadioEvents->TxDone() / RadioEvents->RxError(); / RadioEvents->RxDone / RadioEvents->TxTimeout() / .., and also transmits on SPI ( modify RADIO MODE ).
Those.  from the "LORA" task the functions OnRadioTxDone/ OnRadioRxDone / ... are called.

For example, the OnRadioRxDone function changes the fields of global variables: LoRaMacFlags/McpsConfirm/MlmeConfirm/LoRaMacParams/IsLoRaMacNetworkJoined/ and transmits via SPI (Radio.Sleep())

In turn, from the timer functions OnMacStateCheckTimerEvent/ OnTxDelayedTimerEvent/OnRxWindow1TimerEvent/...
the same global variables McpsConfirm/MlmeConfirm/LoRaMacParams/IsLoRaMacNetworkJoined/... are being read or changed And change RADIO (SPI)

nic2019, 2024-01-29:
The "LORA" task calls functions on interrupts. And calls to timer functions are executed independently. In general, we should consider them as competing tasks.

beegee-tokyo, 2024-01-29:
The LoRa task waits for a semaphore released by the interrupt handler. It is not called directly from the interrupt.
And for task/timer handling we have FreeRTOS running in the background.

nic2019, 2024-01-29:
ok. I put it figuratively.
There are several tasks (FreeRTOS): LoRa task, scheduler-timers , main loop that simultaneously read and change global variables: LoRaMacFlags/McpsConfirm/MlmeConfirm/LoRaMacParams/IsLoRaMacNetworkJoined/   And transmits via SPI (Radio.Sleep()).

In the C++ specification, this is a data race - undefined behavior.

beegee-tokyo, 2024-01-29:
If you have a problem with it, the library is open source and you can improve it.
Many users of this library and no complains so far.

closed 2024-01-29 by beegee-tokyo