beegee-tokyo/SX126x-Arduino #120: OTAA Fails, but network receives join and returns JoinAccept

cburza, 2024-03-04:
I am using a Heltec Wireless Stick Lite(V3). I am unable to get the OTAA working. I have tested with TTN and Helium. Both platforms received the join request and respond with JoinAccept, but the node never successfully joins the network. The hardware definitely works, as I was able to join TTN without much issues using the Heltec Arduino example code that uses their own library and custom ESP32 core code.

I am using the example code [LoRaWanPio](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/examples/LoRaWanPio/src/main.cpp) which I slightly modified to update for the board and region and removed code that is not applicable.

I also tried removing/adding `lmh_setSubBandChannels(1)` which does not seem to make a difference.

The `platformio.ini`.
```
[env:heltec_wifi_lora_32_V3]
platform = espressif32
board = heltec_wifi_lora_32_V3
framework = arduino
lib_deps =
    SX126x-Arduino
monitor_speed = 115200
build_flags = -DLIB_DEBUG=1
```

My `main.cpp`.
```c
#include <Arduino.h>

#include <LoRaWan-Arduino.h>
#include <SPI.h>

#define SCHED_MAX_EVENT_DATA_SIZE APP_TIMER_SCHED_EVENT_DATA_SIZE /**< Maximum size of scheduler events. */
#define SCHED_QUEUE_SIZE 60										  /**< Maximum number of events in the scheduler queue. */

#define LORAWAN_APP_DATA_BUFF_SIZE 64  /**< Size of the data to be transmitted. */
#define LORAWAN_APP_TX_DUTYCYCLE 10000 /**< Defines the application data transmission duty cycle. 10s, value in [ms]. */
#define APP_TX_DUTYCYCLE_RND 1000	   /**< Defines a random delay for application data transmission duty cycle. 1s, value in [ms]. */
#define JOINREQ_NBTRIALS 3			   /**< Number of trials for the join request. */

hw_config hwConfig;

int PIN_LORA_RESET = 12;	 // LORA RESET
int PIN_LORA_NSS = 8;	 // LORA SPI CS
int PIN_LORA_SCLK = 9;	 // LORA SPI CLK
int PIN_LORA_MISO = 11;	 // LORA SPI MISO
int PIN_LORA_DIO_1 = 14; // LORA DIO_1
int PIN_LORA_BUSY = 13;	 // LORA SPI BUSY
int PIN_LORA_MOSI = 10;	 // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE

// Foward declaration
static void lorawan_has_joined_handler(void);
static void lorawan_rx_handler(lmh_app_data_t *app_data);
static void lorawan_confirm_class_handler(DeviceClass_t Class);
static void lorawan_join_failed_handler(void);
static void send_lora_frame(void);
static uint32_t timers_init(void);

// APP_TIMER_DEF(lora_tx_timer_id);                                              ///< LoRa tranfer timer instance.
TimerEvent_t appTimer;														  ///< LoRa tranfer timer instance.
static uint8_t m_lora_app_data_buffer[LORAWAN_APP_DATA_BUFF_SIZE];			  ///< Lora user application data buffer.
static lmh_app_data_t m_lora_app_data = {m_lora_app_data_buffer, 0, 0, 0, 0}; ///< Lora user application data structure.

/**@brief Structure containing LoRaWan parameters, needed for lmh_init()
 */
static lmh_param_t lora_param_init = {LORAWAN_ADR_ON, LORAWAN_DEFAULT_DATARATE, LORAWAN_PUBLIC_NETWORK, JOINREQ_NBTRIALS, LORAWAN_DEFAULT_TX_POWER, LORAWAN_DUTYCYCLE_OFF};

/**@brief Structure containing LoRaWan callback functions, needed for lmh_init()
*/
static lmh_callback_t lora_callbacks = {BoardGetBatteryLevel, BoardGetUniqueId, BoardGetRandomSeed,
                                        lorawan_rx_handler, lorawan_has_joined_handler,
                                        lorawan_confirm_class_handler, lorawan_join_failed_handler};

uint8_t nodeDeviceEUI[8] = {Redacted};

[... 406 lines cut ...]
<LM> OnMacStateCheckTimerEvent
<LM> OnMacStateCheckTimerEvent
<LM> OnMacStateCheckTimerEvent
<BRD> LoRa task wakeup
<RADIO> IRQ_RX_TIMEOUT
<LM> OnRadioRxTimeout
<RADIO> RX window timeout = 3000
<LM> OnMacStateCheckTimerEvent
<LM> OnMacStateCheckTimerEvent
<LM> OnMacStateCheckTimerEvent
<LM> OnMacStateCheckTimerEvent
<BRD> LoRa task wakeup
<RADIO> IRQ_RX_TIMEOUT
<LM> OnRadioRxTimeout
<LM> OnMacStateCheckTimerEvent
<LM> Join network failed 3 time(s)

<LM> LoRaMacState = idle
OVER_THE_AIR_ACTIVATION failed!
Check your EUI's and Keys's!
Check if a Gateway is in range!

```

What could be the problem?

beegee-tokyo, 2024-03-05:
Can you try to add `_hwConfig.USE_RXEN_ANT_PWR = false` to the hardware configuration

```cpp
    // Define the HW configuration between MCU and SX126x
    hwConfig.CHIP_TYPE = SX1262_CHIP;		  // Example uses an eByte E22 module with an SX1262
    hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
    hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	  // LORA SPI CS
    hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;	  // LORA SPI CLK
    hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;	  // LORA SPI MISO
    hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
    hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;	  // LORA SPI BUSY
    hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;	  // LORA SPI MOSI
    hwConfig.RADIO_TXEN = RADIO_TXEN;		  // LORA ANTENNA TX ENABLE
    hwConfig.RADIO_RXEN = RADIO_RXEN;		  // LORA ANTENNA RX ENABLE
    hwConfig.USE_DIO2_ANT_SWITCH = true;	  // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
    hwConfig.USE_DIO3_TCXO = true;			  // Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
    hwConfig.USE_DIO3_ANT_SWITCH = false;	  // Only Insight ISP4520 module uses DIO3 as antenna control
    hwConfig.USE_RXEN_ANT_PWR = false;          // Antenna switch is powered up all the time
```

cburza, 2024-03-05:
@beegee-tokyo I added `hwConfig.USE_RXEN_ANT_PWR = false;`. That does not seem to make a difference.

beegee-tokyo, 2024-03-06:
If you disable the library debug output, does it change anything?
The library debug output is helpful sometimes, but it has as well an influence on timing.

Other than that I have not much ideas. I have ESP32 working fine here ([RAK11200](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK11200/Overview/) with [RAK13300](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK13300/Overview) LoRa module).

cburza, 2024-03-06:
Unfortunately that did not change anything.

beegee-tokyo, 2024-03-06:
Sorry, I cannot help here. I do not have the Heltec stick and the ESP32 hardware I have at hand is working.

closed 2024-04-09 by beegee-tokyo