beegee-tokyo/SX126x-Arduino #122: Issue with porting ESP-IDF Integration for SX1262 Module

ABHITHLALC, 2024-03-19:
Hi @beegee-tokyo 
I am reaching out to seek assistance with an issue I am encountering while attempting to integrate the SX1262 LoRa module with the ESP-IDF framework.
I have been working on porting the SX126x-Arduino library to the ESP-IDF platform for use with an ESP32 microcontroller. While I have successfully ported and built the code, I am experiencing difficulties establishing communication with the network server.
The SX1262 module is failing to join the network server, Despite following the configuration steps outlined in the documentation, including setting the correct SyncWord and channel frequencies, the module fails to establish a connection.

Troubleshooting Steps Taken:

- Verified hardware connections between the ESP32 and the SX1262 module, ensuring correct wiring of SPI and control pins.
- Flashed the Arduino lorawan example code to same hadware setup and it is connected to network server and sending lora frames.
- Ensured stable power supply to the SX1262 module

Despite these efforts, I have been unable to resolve the issue and would greatly appreciate any guidance or assistance your team can provide. If possible, I would appreciate any suggestions for troubleshooting steps or configuration adjustments to achieve successful communication with the network server.

Hardware: ESP32 microcontroller with the Waveshare SX1262 LoRa module. 
Software: ESP-IDF framework version 4.4.1 and SX1262-Arduino library by beegee-tokyo.
Relevant Code Snippets: I can provide excerpts from the ESP-IDF code and configuration files upon request.

The LOG from ESP-IDF code:
```
I (0) cpu_start: App cpu up.
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 160000000
I (259) cpu_start: Application information:
I (263) cpu_start: Project name:     lora_arduino
I (269) cpu_start: App version:      1
I (273) cpu_start: Compile time:     Mar 19 2024 11:02:36
I (279) cpu_start: ELF file SHA256:  9cc270fff4e5255c...
I (285) cpu_start: ESP-IDF:          v4.4.1
I (290) heap_init: Initializing. RAM available for dynamic allocation:
I (297) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (303) heap_init: At 3FFB4A70 len 0002B590 (173 KiB): DRAM
I (310) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (316) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (323) heap_init: At 4008D0D4 len 00012F2C (75 KiB): IRAM
I (330) spi_flash: detected chip: generic
I (333) spi_flash: flash io: dio
I (338) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (2344) SPI: SPI initialized
I (2344) SX1262: SPI and pins initialized
I (2374) SX1262: Node reset
I (2374) BRD: SyncWord = 2414
I (2374) Lora_task: LoRa Task started
I (2374) LMH: OTAA 
DevEui=70-B3-D5-7E-D0-06-4A-9A
DevAdd=00000000
AppEui=00-00-00-00-00-00-00-00
AppKey=88-67-44-AB-01-06-F9-D1-82-28-72-DB-69-4D-40-E1
I (2414) SX1262: Node reset
I (2414) IRQ_PIN: Lora IRQ initialising
I (2414) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:1 
W (2434) SX1262: [SX126xWaitOnBusy] Timeout waiting for BUSY pin to be low
W (2444) SX1262: [SX126xWaitOnBusy] Timeout waiting for BUSY pin to be low
I (2444) LMH: Selected subband 1
I (2454) LMH: Selected subband 1
I (2454) EU868: Channel count ch# 0, freq 868100000
I (2454) REGION: EU868 , Set channel 0, frequency 868100000
I (2464) EU868: Channel count ch# 1, freq 868300000
I (2464) REGION: EU868 , Set channel 1, frequency 868300000
I (2474) EU868: Channel count ch# 2, freq 868500000
I (2474) REGION: EU868 , Set channel 2, frequency 868500000
I (2484) EU868: Channel count ch# 3, freq 0
I (2484) EU868: Channel count ch# 4, freq 0
I (2494) EU868: Channel count ch# 5, freq 0
I (2494) EU868: Channel count ch# 6, freq 0
I (2504) EU868: Channel count ch# 7, freq 0
I (2504) REGION: EU868, Using channel 0, frequency 868100000
I (2514) LM: dutyCycleTimeOff was = 0
I (2514) LM: Set send_join_now to false
W (2524) RADIO: <<<Trying for Send Payload>>>
W (2524) RADIO: <<<Setting up TX mode>>>
I (2534) LORA: Did not join network, skip sending frame
I (2634) LM: OnMacStateCheckTimerEvent
I (2884) LM: OnMacStateCheckTimerEvent
I (3134) LM: OnMacStateCheckTimerEvent
I (3384) LM: OnMacStateCheckTimerEvent
I (3634) LM: OnMacStateCheckTimerEvent
I (3884) LM: OnMacStateCheckTimerEvent
I (4134) LM: OnMacStateCheckTimerEvent
I (4384) LM: OnMacStateCheckTimerEvent
I (4634) LM: OnMacStateCheckTimerEvent
I (4884) LM: OnMacStateCheckTimerEvent
I (5134) LM: OnMacStateCheckTimerEvent
I (5384) LM: OnMacStateCheckTimerEvent
W (5534) RADIO: Inside the radio irq
I (5534) RADIO: TimerTxTimeout
I (5534) LM: OnRadioTxTimeout 
W (5544) SX1262: [SX126xWaitOnBusy] Timeout waiting for BUSY pin to be low
W (5554) SX1262: [SX126xWaitOnBusy] Timeout waiting for BUSY pin to be low
I (5634) LM: OnMacStateCheckTimerEvent
I (5634) LM: LoRaMacState = idle
I (5634) LORA: OVER_THE_AIR_ACTIVATION failed!
I (5634) LORA: Check your EUI's and Keys's!   
I (5644) LORA: Check if a Gateway is in range!
```

Thank you for your attention to this matter. I am looking forward to your prompt assistance in resolving this issue and achieving successful integration of the SX1262 module with the ESP-IDF platform.

beegee-tokyo, 2024-03-20:
There could be a problem with the connection to between the ESP32 and the SX1262
```log
W (2434) SX1262: [SX126xWaitOnBusy] Timeout waiting for BUSY pin to be low
```
Says that BUSY pin of the SX1262 is never seen to go low. Either the pin assignment is wrong or the power supply of the SX1262 is wrong.

Another thing is AppEUI all zeros. While TTN states that this is ok, it is not good to do it and I have seen problems with other LNS when the AppEUI is all zeros.

Third thing, do you see the join request packet on the gateway log or on the LNS log?

ABHITHLALC, 2024-03-20:
> Says that BUSY pin of the SX1262 is never seen to go low. Either the pin assignment is wrong or the power supply of the SX1262 is wrong.

This is the logic analyzer screenshot. which shows the BUSY pin state changes. But the DI01 pin(ISR) is always low.
[image]

> Another thing is AppEUI all zeros. While TTN states that this is ok, it is not good to do it and I have seen problems with other LNS when the AppEUI is all zeros.

Yes, It is not good to have all zero's in AppEUI. But this is for testing purpose.  I've sent LoRa messages to the same TTN device with the same credentials using Arduino code.

> Third thing, do you see the join request packet on the gateway log or on the LNS log?

No, there is nothing on Gateway Log and the TTN console

beegee-tokyo, 2024-03-20:
EU868 is what you want to use? That's what I see in the logs.
I don't see a TX finished IRQ callback in the logs, only TX timeout. Do you call lmh_join()?

ABHITHLALC, 2024-03-20:
Yes, I am calling lmh_join() in main.c
```
void app_main()
{
	uint32_t ret;
	
	
	ret = lora_hardware_init();
	if(ret != ESP_OK)
	{
		ESP_LOGE("LORA","Lora hardware init failed - %d", ret);
	}

	ret = timers_init();
	if(ret != ESP_OK)
	{
		ESP_LOGE("LORA","timers_init failed - %d", ret);
	}

    lmh_setDevEui(nodeDeviceEUI);
	lmh_setAppEui(nodeAppEUI);
	lmh_setAppKey(nodeAppKey);

    ret = lmh_init(&lora_callbacks, lora_param_init, true, CLASS_A, LORAMAC_REGION_EU868);
	if(ret != ESP_OK)
	{
		ESP_LOGE("LORA","Lmh initialisation failed init failed - %d", ret);
	}

	if (!lmh_setSubBandChannels(1))
	{
		ESP_LOGE("LORA","lmh_setSubBandChannels failed. Wrong sub band requested?");
	}

	/*Joining process starts*/
    lmh_join();
	send_lora_frame();
	
}
```

beegee-tokyo, 2024-03-20:
If you do not see packets in the Gateway, it could be the antenna switch control is not correct, the antenna is not connected correct or damaged.
Difficult to say.
Do you have an SDR to check whether there is really a signal sent by the device?

ABHITHLALC, 2024-03-21:
Sorry I don't have SDR. Is there any other methods to test and verify?

beegee-tokyo, 2024-03-21:
You could try to measure the power consumption of the device. You should see a significant peak in the current (~100mA) while the SX1262 is transmitting.
But for that you need a good instrument as well, like the PPK2 (Nordic Power Profile Kit II)

UI-Jakob, 2024-04-03:
Would you be able to share your port to the ESP-IDF? I have the same LoRa module as you and I am trying to get it to work with an ESP32 C6.

beegee-tokyo, 2024-04-03:
I am sorry, I have no experience with ESP-IDF, I never used it.
I cannot help with the porting.

UI-Jakob, 2024-04-03:
My comment was more to @ABHITHLALC :) Perhaps I can help him with the porting if he like to share his current progress.

beegee-tokyo, 2024-04-03:
Maybe these help:

[Arduino-Library-with-IDF](https://github.com/sukesh-ak/Arduino-Library-with-IDF)
[Best approach today to running Arduino libraries with ESP-IDF?](https://community.platformio.org/t/best-approach-today-to-running-arduino-libraries-with-esp-idf/35608)

closed 2024-04-09 by beegee-tokyo