beegee-tokyo/SX126x-Arduino #123: Gateway traffic shows "Join Request" -> "Join Accept" but the LoRa end device says OVER_THE_AIR_ACTIVATION failed!

theanhoo, 2024-03-29:
I am new to LoRa. 

I have a LILYGO T-BEAM v1.1 with an SX1262 connecting to a Dragino LPS8-V2 gateway, attempting to join The Things Stack.

During an OTAA attempt, The Things Stack shows:
```
Receive join-request
Send join-request to cluster-local ...
Accept join-request
Join-request to cluster-local Join...
Successfully processed join-request
Schedule join-accept for transmission...
Successfully scheduled join-accept...
Receive join-accept message
Forward join-accept message
Forward join-accept to Application Server...
Downlink successfully transmitted
```
Dragino gateway shows:
```
Join Request (uplink)
Join accept (downlink)
```
However, the Arduino IDE shows the following in Serial Monitor:
```
=====================================
SX126x LoRaWan test
=====================================
OVER_THE_AIR_ACTIVATION failed!
Check your EUI's and Keys's!
Check if a Gateway is in range!
```

The error shows up approximately 3.3 seconds after the test starts. Perhaps it's a timeout issue?

Please advise.

Thank you.

beegee-tokyo, 2024-03-29:
Check the RX1 and RX2 downlink frequencies. If they are different in the gateway and in TTS, the join accept is not received by the node.

What LoRaWAN region?

theanhoo, 2024-03-29:
AS923-2

Thanks. I'll check.

illperipherals, 2024-03-29:
I have also seen cases where the node RSSI is too close to the limit to get that downlink packet... I would check your antenna  and RSSI values as well.

theanhoo, 2024-03-30:
Thanks. I will check RSSI, as well.

theanhoo, 2024-03-30:
[image]

The list below shows frequencies on the Dragino LPS8-V2 gateway for `AS923 Asia AS923-1 (920-923MHz)`
```
Gateway Channels Frequnecy
---------------------------------------
chan_multSF_0
Lora MAC, 125kHz, all SF, 923.2 MHz
---------------------------------------
chan_multSF_1
Lora MAC, 125kHz, all SF, 923.4 MHz
---------------------------------------
chan_multSF_2
Lora MAC, 125kHz, all SF, 922.2 MHz
---------------------------------------
chan_multSF_3
Lora MAC, 125kHz, all SF, 922.4 MHz
---------------------------------------
chan_multSF_4
Lora MAC, 125kHz, all SF, 922.6 MHz
---------------------------------------
chan_multSF_5
Lora MAC, 125kHz, all SF, 922.8 MHz
---------------------------------------
chan_multSF_6
Lora MAC, 125kHz, all SF, 923.0 MHz
---------------------------------------
chan_multSF_7
Lora MAC, 125kHz, all SF, 922.0 MHz
---------------------------------------
chan_Lora_std
Lora MAC, 250kHz, SF7, 922.1 MHz
---------------------------------------
chan_FSK
FSK 50kbps, 921.8 MHz
---------------------------------------
```

I have some questions.

1. Are RX1, RX2 and RSSI controlled by the regional setting or are the values set explicitly in code?
2. I have chosen `AS923 Asia AS923-1 (920-923MHz)` on the gateway and `LORAMAC_REGION_AS923` in the code. Do they match?
3. The gateway shows two entries of AS923-1 but there is no corresponding enum in `eLoRaMacRegion_t`. Is that by design?

Please advise.

Thanks.

```
typedef enum eLoRaMacRegion_t
{
	/*!
     * AS band on 923MHz
     */
	LORAMAC_REGION_AS923, <--- Using this
	/*!
     * Australian band on 915MHz
     */
	LORAMAC_REGION_AU915,
	/*!
     * Chinese band on 470MHz
     */
	LORAMAC_REGION_CN470,
	/*!
     * Chinese band on 779MHz
     */
	LORAMAC_REGION_CN779,
	/*!
     * European band on 433MHz
     */
	LORAMAC_REGION_EU433,
	/*!
     * European band on 868MHz
     */
	LORAMAC_REGION_EU868,
	/*!
     * South korean band on 920MHz
     */
	LORAMAC_REGION_KR920,
	/*!
     * India band on 865MHz
     */
	LORAMAC_REGION_IN865,
	/*!
     * North american band on 915MHz
     */
	LORAMAC_REGION_US915,
	/*!
     * AS band on 923MHz -1.8MHz
     */
	LORAMAC_REGION_AS923_2,
	/*!
     * AS band on 923MHz -6.6MHz
     */
	LORAMAC_REGION_AS923_3,
	/*!
     * AS band on 923MHz -5.9MHz
     */
	LORAMAC_REGION_AS923_4,
	/*!
     * Russia band on 864MHz
     */
	LORAMAC_REGION_RU864,
} LoRaMacRegion_t;
```

beegee-tokyo, 2024-03-30:
AS923 and AS923-1 are the same, 1,2,3 and 4 variants were added when new regions were adopted.

In node library the settings for the region are in [RegionAS923.cpp](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/mac/region/RegionAS923.cpp) and they work for me. I am using AS923-3, which is the same as AS923-1 with a frequency shift of 6.6MHz.

RX1 window frequency is the same as the frequency that the node used for the uplink and RX2 window frequency is fixed to 923.3MHz (uint32_t AS923_RX_WND_2_FREQ = 923200000;)[https://github.com/beegee-tokyo/SX126x-Arduino/blob/98ef65b5a2e9c35eabfd80a466327a9fdbb8e38e/src/mac/region/RegionAS923.cpp#L44]

I don't know about the settings in the Dragino gateway or why there are two AS923-1 variants.

On TheThingsStack, for the gateway I am setting AS923 Group 3, you should use AS923 Group 1:
[image]

I am using RAK WisGate Edge gateways and my settings there are 
[image]

But for AS923-1 I would choose
[image]

Maybe you can check in the Dragino settings what frequencies are used.

# ---------------------------------

Last not least, a problem I have seen before. In the RF path from the SX1262 to the antenna is a antenna switch.
It needs to be setup properly in the connection setup or the SX1262 might not be able to receive packets.

I do not know how the antenna switch is controlled in the LILYGO T-BEAM v1.1 with an SX1262, there are multiple options.
e.g. on the RAK4631, the antenna switch direction is controlled by DIO2 of the SX1262 and the power for the antenna switch is controlled by an GPIO of the MCU.

You should check how the Liligo MCU is controlling the antenna switch in the schematics and make sure the hardware configuration is correct
[image]

theanhoo, 2024-04-01:
It seems to be working now. Thank you.
```
=====================================
SX126x LoRaWan test
=====================================
OVER_THE_AIR_ACTIVATION != 0
Sending frame
lmh_send result 0
Sending frame
lmh_send result 0
Sending frame
lmh_send result 0
Sending frame
lmh_send result 0
Sending frame
lmh_send result 0
```

These are the pins set for LILYGO T-BEAM v1.1 LORA32 SX1262:
```
int PIN_LORA_RESET = 23;	// LORA RESET
int PIN_LORA_NSS = 18;		// LORA SPI CS
int PIN_LORA_SCLK = 5;		// LORA SPI CLK
int PIN_LORA_DIO_1 = 33;	// LORA DIO_1
int PIN_LORA_BUSY = 32;		// LORA SPI BUSY
int PIN_LORA_MISO = 19;		// LORA SPI MISO
int PIN_LORA_MOSI = 27;		// LORA SPI MOSI
int RADIO_TXEN = -1;		// LORA ANTENNA TX ENABLE
int RADIO_RXEN = 26;		// LORA ANTENNA RX ENABLE
```

However, I did not change the region as 925-928MHz on the Dragino LPS8-V2 gateway is labeled `AU915 Australia 915MHz (915~928)` and I am not in Australia.

Again, thank you.

beegee-tokyo, 2024-04-01:
Glad it works now.
I guess it was the antenna switch control.

closed 2024-04-01 by beegee-tokyo