beegee-tokyo/SX126x-Arduino #125: Connection Issue with SX1262 LoRa Node on Australian Frequency (AU915)

ABHITHLALC, 2024-04-25:
I am experiencing a connection issue with the SX1262 LoRa node when attempting to connect to The Things Network (TTN) using the sx126x-Arduino library. The device connects and sends LoRa packets successfully on EU868, but when I test with AU915 frequency band, the device is not showing any uplink(even joining request). Here's a detailed description of the issue and the steps I followed to reproduce it.

Hardware: SX1262 LoRa node with ESP32
Gateway frequency: 915 MHz
Region: Australia
Development setup: ( Arduino platform with ESP32)
The Things Network (TTN) setup: Cluster address -au1.cloud.thethings.network
                                                    : Frequency plan -AU_915_928_FSB_2

Steps to Reproduce:
Configure the node to use the AU915 frequency band.
Attempt to connect to TTN through a 915 MHz gateway.
Observe that the device does not connect to TTN.

Expected Behavior:
The SX1262 LoRa node should successfully connect to TTN on the AU915 frequency band and be able to send LoRa packets.

Actual Behavior:
The device shows "never connected" when attempting to connect to TTN using the AU915 frequency band.
Is there a known issue with the beegetokyo library when using the AU915 frequency band?
Are there any specific configurations or settings required to work with TTN on the AU915 frequency?
Could this be a hardware compatibility issue?

beegee-tokyo, 2024-04-26:
Did you set the subband to subband 2?
AU_915_928_FSB_2 ==> FSB_2 means subband 2
[Specify Subbands](https://github.com/beegee-tokyo/SX126x-Arduino?tab=readme-ov-file#specifiy-sub-bands)

ABHITHLALC, 2024-05-02:
My gateway is a Raspberry Pi with the Waveshare SX1302 915MHz module. It's active on The Things Network (TTN), and I've set it up for AU915 Sub-Band 2.

I resolved my problem by changing the LoRaWAN version and regional parameters from 1.0.4 back to 1.0.3 on the TTN device. Now device is active and sending the lora frames to the TTN. And can be seen in TTN console/

I'm curious if the library supports LoRaWAN Specification 1.0.4 and RP002 Regional Parameters 1.0.4. Can you confirm this?

beegee-tokyo, 2024-05-02:
Your question is answered in the first 5 lines of the README.MD of this repo.

[image]

closed 2024-05-02 by beegee-tokyo

ABHITHLALC, 2024-05-02:
Sorry I missed that. Thanks for pointing it out. I'll take a look at the README.MD, and I appreciate you directing me there. My apologies for the oversight.