beegee-tokyo/SX126x-Arduino #126: ESP32 + SX1262(Waveshare) is not working, but it is working with Radiolib library

Jaishankar872, 2024-05-10:
Previously I am using the Radiolib library with below setting, it is working fine

With Radiolib

SPIClass spi(HSPI);
//HPSI -> SCLK: 14, MOSI: 13, MISO: 12
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);
  SX1262 radio = new Module(15, 2, 17, 16, spi, spiSettings);
//SX1262 radio = new Module(cs, irq, rst, gpio, spi, spiSettings);
// NSS pin: 15 (CS)
// DIO1 pin: 2
// NRST pin: 17
// BUSY pin: 16

void setup() {
  Serial.begin(9600);
  spi.begin();
*******************************************************************************************************************

Now for the same board, I tried using this SX126x-Arduino with below setting, but it is not working throwing timeout error

#ifdef ESP32
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 17;  // LORA RESET
int PIN_LORA_DIO_1 = 2; // LORA DIO_1
int PIN_LORA_BUSY = 16;  // LORA SPI BUSY
int PIN_LORA_NSS = 15;	// LORA SPI CS
int PIN_LORA_SCLK = 14;  // LORA SPI CLK
int PIN_LORA_MISO = 12;  // LORA SPI MISO
int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE

Serial output
=====================================
SX126x PingPong test
=====================================
MCU Espressif ESP32
BoardId: 00-00-C8-34-13-B3-A3-A0
Starting lora_hardware_init
<BRD> SyncWord = 2414
<BRD> LoRa Task started
Starting Radio.Rx
<RADIO> RX window timeout = 3000
<RADIO> TimerRxTimeout
OnRxTimeout

***********************************************************************************************************************************
**Please help here to find & fix the error**

Thanks in advance!!!

beegee-tokyo, 2024-05-11:
```cpp
int RADIO_TXEN = -1; // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1; // LORA ANTENNA RX ENABLE
```
Usually does not work. I don't know the ESP32 - Waveshare module, but on most boards you need to control the power and/or the direction of the RF switch in the path from the SX1262 to the antenna connector.

Another option (in case RadioLib does some magic on the RF switch control) is to increase the RX timeout to 5000 milliseconds. There seems to be some problem in the latest version of this library with the timeouts. But I didn't find time to check it out.

Jaishankar872, 2024-05-11:
1. Is there any way to conform the connection between the ESP32 and SX1262 module working? [In RadioLib check function is available]
2. If there is problem in latest version, will the old version help?
3. I am know try your library's PingPong Example with two identical devices, which is not working. When the transmission from the lora will happen?(I read both code and Readme.md which doesn't help for me understand)

beegee-tokyo, 2024-05-11:
From your log
```log
Starting lora_hardware_init
SyncWord = 2414
```
Tells me the SPI connection is working.

There seem to be a problem with RX timeouts in the latest version. I do not see that problem, it seems to be related to the way examples are setting the timeouts.

I have to check how the PingPong works by myself. I wrote that example 3 years ago. I was not using that example for a long time.

beegee-tokyo, 2024-05-11:
Btw, if you are using LoRa P2P, the RadioLib library is easier to use than this library.
Why do you need to change?

Jaishankar872, 2024-05-11:
@beegee-tokyo 
1. 
Now I tested the SPI is connection is fine
*************************************************************************************************************************
**Manually I added the Code:** 

	// Initialize the LoRa chip
	Serial.println("Starting lora_hardware_init");
    uint32_t err_code = lora_hardware_init(hwConfig);
	if (err_code != 0)
	{
		Serial.printf("lora_hardware_init failed - %d\n", err_code);
		Serial.printf("Stopped the funtion");
		while (1);		
	}else{
		Serial.printf("lora_hardware_init Passed.......\n ");
	}
	
	
**Serial Output:**

> 
> SX126x PingPong test
> MCU Espressif ESP32
> BoardId: 00-00-C8-34-13-B3-A3-A0
> Starting lora_hardware_init
> lora_hardware_init Passed.......
>  Starting Radio.Rx
> Get Radio Status Radio..........
> OnRxTimeout

*************************************************************************************************************************

2.
    P2P is not required for me, Actually I started this library with PingPong Example as start to understand how this library function.
    But I unable to get the result mentioned in that example Readmd File

Jaishankar872, 2024-05-11:
My Final Idea is to build the home automation product as
Single Channel LoRaWAN Gateway (https://github.com/beegee-tokyo/SX1262-SC-GW)  or 
Add with ESPhome (https://github.com/PaulSchulz/esphome-lora-sx126x)

So your support will greatly helps me!!!

beegee-tokyo, 2024-05-12:
For the single channel gateway, it is not recommended anymore to use this type of gateways. In addition, I am not sure if the code still works. It is a long time since I created and used it. Today I am always using 8 channel gateways, which are now available for a reasonable price.

For the ESPhome integration, you need to get LoRa P2P to work, as this is what will be used.

For next steps, I would propose to go back to library version 2.0.23 and check if it works or try to change the RX window timeout to a larger value.

Jaishankar872, 2024-05-12:
Thanks for your suggestion 
I will go with ESPhome Integration (https://github.com/PaulSchulz/esphome-lora-sx126x)

To move forward, I want understand the send & receiving data with this library to use this [esphome-lora-sx126x] Library

As your suggestion, now I want back to library version 2.0.23, with large RX time windows(60secs)

**Serial Output with PingPong Example Code**
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13232
> load:0x40080400,len:3028
> entry 0x400805e4
> Runon on the library: beegee-tokyo/SX126x-Arduino@^2.0.23
> SX126x PingPong test
> MCU Espressif ESP32
> BoardId: 00-00-C8-34-13-B3-A3-A0
> Starting lora_hardware_init
> <BRD> SyncWord = 2414
> lora_hardware_init Passed.......
>  <BRD> LoRa Task started
> Starting Radio.Rx
> <RADIO> RX window timeout = 60000
> AddExtra Setup Endline Reached -------------------
> <RADIO> TimerRxTimeout
> OnRxTimeout

But I unable to get the result mentioned in that example Readmd File.

This library also not working (https://github.com/PaulSchulz/esphome-lora-sx126x) unable send or receive the lora

[image]

Please help here to understand the sending and receive data, I believe this PingPong Example code will help

beegee-tokyo, 2024-05-12:
I am sorry, I don't know the LoRa module you are using, and I am not sure what would be the correct setup for it.
Obviously you have the SPI connection correct, but there seems something to be missing.

I don't know the ESPhome library either, but I am guessing it is the same problem.

Do you have schematics or other information about the "ESP32 + SX1262 Waveshare" setup you are using?

Jaishankar872, 2024-05-12:
Yes I have the schematics

**ESP32 and Lora Module schematics**

[image]
                            |
[image]

**Lora Module SX1262(Waveshare) Schematics**

[image]

beegee-tokyo, 2024-05-13:
RX_EN and TX_EN are not connected, that means your RF switch U4 is neither powered nor has any direction control.

That cannot work. This two pins must be connected to something.

Options 
- connect RX_EN to 3.3V and TX_EN to DIO2.
    - hwConfig.USE_DIO2_ANT_SWITCH = hwConfig.USE_DIO2_ANT_SWITCH; // LORA DIO2 controls antenna
- connect RX_EN and TX_EN to GPIO's of the ESP32
    - hwConfig.RADIO_TXEN = GPIO of the ESP32;					  // LORA ANTENNA TX ENABLE (e.g. eByte E22 module)
    - hwConfig.RADIO_RXEN = GPIO of the ESP32;					  // LORA ANTENNA RX ENABLE (e.g. eByte E22 module)

Jaishankar872, 2024-05-14:
Connect RX_EN to GPIO18 & TX_EN to DIO2.
[image]

 I done this hardware change and in code added below line

	pinMode(18, OUTPUT);
	digitalWrite(18, HIGH);
        hwConfig.USE_DIO2_ANT_SWITCH = hwConfig.USE_DIO2_ANT_SWITCH; // LORA DIO2 controls antenna

**Full Code Used** 

	#ifdef ESP32
	// ESP32 - SX126x pin configuration
	int PIN_LORA_RESET = 17;  // LORA RESET
	int PIN_LORA_DIO_1 = 2; // LORA DIO_1
	int PIN_LORA_BUSY = 16;  // LORA SPI BUSY
	int PIN_LORA_NSS = 15;	// LORA SPI CS
	int PIN_LORA_SCLK = 14;  // LORA SPI CLK
	int PIN_LORA_MISO = 12;  // LORA SPI MISO
	int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
	int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
	int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE
	#endif
        void setup()        
        {
	pinMode(LED_BUILTIN, OUTPUT);
	digitalWrite(LED_BUILTIN, LOW);
	pinMode(18, OUTPUT);
	digitalWrite(18, HIGH);
	// Define the HW configuration between MCU and SX126x
	hwConfig.CHIP_TYPE = SX1262_CHIP;		  // Example uses an eByte E22 module with an SX1262
	hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
	hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	 // LORA SPI CS
	hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // LORA SPI CLK
	hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // LORA SPI MISO
	hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
	hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // LORA SPI BUSY
	hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // LORA SPI MOSI
	hwConfig.RADIO_TXEN = RADIO_TXEN;		  // LORA ANTENNA TX ENABLE
	hwConfig.RADIO_RXEN = RADIO_RXEN;		  // LORA ANTENNA RX ENABLE
	hwConfig.USE_DIO2_ANT_SWITCH = hwConfig.USE_DIO2_ANT_SWITCH; // LORA DIO2 controls antenna //true;	  // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
	hwConfig.USE_DIO3_TCXO = true;			  // Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
	hwConfig.USE_DIO3_ANT_SWITCH = false;	 // Only Insight ISP4520 module uses DIO3 as antenna control

	// Initialize Serial for debug output
	Serial.begin(115200);

	Serial.println("Runon on the library: beegee-tokyo/SX126x-Arduino@^2.0.23");
	Serial.println("SX126x PingPong test");

        #ifdef NRF52_SERIES
	Serial.println("MCU Nordic nRF52832");
	pinMode(30, OUTPUT);
	digitalWrite(30, HIGH);
	// Start BLE if we compile for nRF52
	initBLE();
        #endif
        #ifdef ESP32
	Serial.println("MCU Espressif ESP32");
        #endif
        #ifdef ESP8266
	Serial.println("MCU Espressif ESP8266");
        #endif

	uint8_t deviceId[8];

	BoardGetUniqueId(deviceId);
	Serial.printf("BoardId: %02X-%02X-%02X-%02X-%02X-%02X-%02X-%02X\n",
				  deviceId[7],
				  deviceId[6],
				  deviceId[5],
				  deviceId[4],
				  deviceId[3],
				  deviceId[2],
				  deviceId[1],
				  deviceId[0]);

	// Initialize the LoRa chip
	Serial.println("Starting lora_hardware_init"); 
         uint32_t err_code = lora_hardware_init(hwConfig);
	if (err_code != 0)
	{
		Serial.printf("lora_hardware_init failed - %d\n", err_code);
		Serial.printf("Stopped the funtion");
		while (1);		
	}else{
		Serial.printf("lora_hardware_init Passed.......\n ");
	}

	// Initialize the Radio callbacks
	RadioEvents.TxDone = OnTxDone;
	RadioEvents.RxDone = OnRxDone;
	RadioEvents.TxTimeout = OnTxTimeout;
	RadioEvents.RxTimeout = OnRxTimeout;
	RadioEvents.RxError = OnRxError;
	RadioEvents.CadDone = OnCadDone;

	// Initialize the Radio
	Radio.Init(&RadioEvents);

	// Set Radio channel
	Radio.SetChannel(RF_FREQUENCY);

	// Set Radio TX configuration
	Radio.SetTxConfig(MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
					  LORA_SPREADING_FACTOR, LORA_CODINGRATE,
					  LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
					  true, 0, 0, LORA_IQ_INVERSION_ON, TX_TIMEOUT_VALUE);

	// Set Radio RX configuration
	Radio.SetRxConfig(MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
					  LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
					  LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
					  0, true, 0, 0, LORA_IQ_INVERSION_ON, true);

	// Start LoRa
	Serial.println("Starting Radio.Rx");
	Radio.Rx(RX_TIMEOUT_VALUE);

    Serial.println("AddExtra Setup Endline Reached -------------------");
	timeToSend = millis();
        }

**Serial Output with PingPong Example Code**
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13232
> load:0x40080400,len:3028
> entry 0x400805e4
> Runon on the library: beegee-tokyo/SX126x-Arduino@^2.0.23
> SX126x PingPong test
> MCU Espressif ESP32
> BoardId: 00-00-C8-34-13-B3-A3-A0
> Starting lora_hardware_init
> <BRD> SyncWord = 2414
> lora_hardware_init Passed.......
>  <BRD> LoRa Task started
> Starting Radio.Rx
> <RADIO> RX window timeout = 10000
> AddExtra Setup Endline Reached -------------------
> <RADIO> TimerRxTimeout
> OnRxTimeout

beegee-tokyo, 2024-05-15:
Sorry, I have no other ideas what could be wrong.
LoRa P2P works without problems for me.

Jaishankar872, 2024-05-15:
Atleast if you provide some idea on problem coming in system my from the above log, helps me to move on step on debug.
Is it possible you share to the difference between RadioLib and SX126x-Arduino Library on P2P communication alone?

beegee-tokyo, 2024-05-16:
I can't give you any more ideas, I have no more ideas.

I do not know RadioLib, if you want to know the differences, you have to compare by yourself.

PaulSchulz, 2024-05-17:
> Sorry, I have no other ideas what could be wrong. LoRa P2P works without problems for me.

Are you referring to regular LoRa radio comms, or is P2P something specific?

beegee-tokyo, 2024-05-17:
I mean your specific problems with the Waveshare device.

beegee-tokyo, 2024-06-05:
Closed, cannot verify problem, no matching hardware available.

closed 2024-06-05 by beegee-tokyo

Jaishankar872, 2024-11-21:
@OtarieSupreme As of now, I didn't any solution in ESPHome side.

But as a work around, I am using the tasmota with same Hardware. Refer [link](https://tasmota.github.io/docs/LoRa-and-LoRaWan-Bridge/)