beegee-tokyo/SX126x-Arduino #127: lmh_send_blocking with odd behaviour

gabrielbravov, 2024-06-17:
Hello Mr. Beegee Tokyo.

We discover a odd behaviour when we use lmh_send_blocking function. With 40000 timeout value, the function. recognizes when i turn off the gateway, but with 100000 timeout value, the function don't recognizes it. This it's the function that i'm using:
```

 lmh_error_status result;
			result = lmh_send_blocking(&m_lora_app_data, LMH_CONFIRMED_MSG, TIMEOUTBLOCKING);
      // result = lmh_send(&m_lora_app_data, LMH_UNCONFIRMED_MSG);

      			#ifndef MAX_SAVE
            Serial.print("el valor de result es: "); Serial.println(result);
            #endif

			switch (result)
			{
			case LMH_SUCCESS:
				    #ifndef MAX_SAVE
            Serial.println("enviado exitosamente (lora handler) caso1");
            #endif
				break;
			case LMH_BUSY:
				    #ifndef MAX_SAVE
            Serial.println("lmh_busy (lora handler)");
            #endif
				break;
			case LMH_ERROR:

        #ifndef MAX_SAVE
           Serial.println("lmh_error, no hay GW o Chirpstack (lora handler) caso2");
        #endif

        // lmh_join();

        break;

			}
```

beegee-tokyo, 2024-06-18:
IMHO, using the blocking send function is the worst thing to do. It is only in the library because a user wanted it. I never fully tested it and that user moved on to use the TX callback functions.

It is much saver to use the TX callback functions to get notified when the TX cycle is finished and to know whether the device received an ACK from the server when using confirmed packet mode.

Without confirmed packet mode, the end node will never know if the gateway or LNS is down.

gabrielbravov, 2024-06-18:
> use the TX callback functio

Thanks for the answer, i read the readme from the library and says that "lorawan_confirmed_finished" callback has a paramter that tells if a confirmation (ACK) was received from the LoRaWAN server or not, but it hasnt examples to know which parameter its and where i find it. ¿can you share some examples to use this callback? i searched in RAK Forums and Github but i didnt find working examples to use this very useful callback.

Thanks in advance,
Gabriel Bravo Valenzuela.

beegee-tokyo, 2024-06-19:
Two callbacks, one for unconfirmed and one for confirmed

https://github.com/beegee-tokyo/SX126x-Arduino/blob/98ef65b5a2e9c35eabfd80a466327a9fdbb8e38e/examples/LORAWAN-OTAA-esp32-e22/src/main.cpp#L249

```cpp

/**
 * @brief Called after unconfirmed packet was sent
 * 
 */
static void lorawan_unconfirm_tx_finished(void)
{
	Serial.println("Uncomfirmed TX finished");
}

/**
 * @brief Called after confirmed packet was sent
 * @param result Result of sending true = ACK received false = No ACK
 */
static void lorawan_confirm_tx_finished(bool result)
{
	Serial.printf("Comfirmed TX finished with result %s", result ? "ACK" : "NAK");
}
```

gabrielbravov, 2024-06-19:
Thanks friend, it's worked flawlessly.

To add more information to other users that has similar problems, i need to add both callbacks to make it work: "lorawan_confirm_tx_finished" and "lorawan_unconfirm_tx_finished"

closed 2024-06-20 by beegee-tokyo