beegee-tokyo/SX126x-Arduino #130: Question - OVER_THE_AIR_ACTIVATION macro?

netmaniac, 2024-11-08:
Hi!
In LoRaWAN example is being used OVER_THE_AIR_ACTIVACTION: https://github.com/beegee-tokyo/SX126x-Arduino/blob/943d4ef24d3de75d6ba45507c6593c23ec7b01c4/examples/LoRaWan/LoRaWan.ino#L244 and I can not find where it is defined?
Only occurrences  are in examples...

beegee-tokyo, 2024-11-09:
In [lmh_init](https://github.com/beegee-tokyo/SX126x-Arduino/tree/943d4ef24d3de75d6ba45507c6593c23ec7b01c4?tab=readme-ov-file#initialize) it is the third parameter.

closed 2024-11-09 by beegee-tokyo

netmaniac, 2024-11-09:
OK, I know that 3rd parameter to lmh_init selects between OTAA/ABP, but this is in runtime. 

My question is about this macro name OVER_THE_AIR_ACTIVATION used in this linked example. Since it is part of #if directive, then it is evaluated at compile time, not run time. I can not find where it is defined - or I'm just something getting wrong?

beegee-tokyo, 2024-11-09:
The macro is not used. It is hardcoded in the lmh_init.