beegee-tokyo/SX126x-Arduino #134: Error in code for function src/mac/LoRaMacHelper.cpp

JDH023, 2024-12-04:
In the function lmh_send_blocking() the code always fails because the function lmh_send() never sets the variable lmh_mac_is_busy to false again when finishing.

Line 1021:
if (LoRaMacMcpsRequest(&mcpsReq) == LORAMAC_STATUS_OK)
{
	lmh_mac_is_busy = true;
	return LMH_SUCCESS;
}

It did not change the variable to false but the result is success so in the fuction lmh_send_blocking() line 1037 it always keeps in the while and finally fails when timeout.

beegee-tokyo, 2024-12-04:
I recommend _**NOT**_ to user the blocking send version. It was implemented a long time ago on request of a single user.

It is better to use the non-blocking send call and the callbacks to handle the results.

JDH023, 2024-12-05:
Hello.
Ok, clear, so better we show some comment over the function to advice it shall not be used.
Anyway it does not help on what I need.
Thanks.

JDH023, 2024-12-05:
Closing issues as it is clear.
Thanks.

closed 2024-12-05 by JDH023