beegee-tokyo/SX126x-Arduino #135: How to properly handle the confirmed frames?

JDH023, 2024-12-04:
Hello.

I am sending confirmed frames to the server and even with the gateway power off the result of the lmh_send() is LHM_SUCCESS.

How I can check that the frame was properly received?

beegee-tokyo, 2024-12-04:
There are two callback functions, one for unconfirmed packets and one for confirmed packets. The second one gives the result as parameter

```cpp
/** LoRaWAN callback after class change request finished */
static void lpwan_unconfirm_tx_finished(void);
/** LoRaWAN callback after class change request finished */
static void lpwan_confirm_tx_finished(bool result);
```

JDH023, 2024-12-05:
Hello.
As result you mean if the frame has been confirmed by the server?
Now I am using lmh_send() and it sayd "LMH_SUCCESS" but it only means the chip did it parts but not confirming if the frame arrived the other side, neither for confirmed frames.
I am going to try and see.

beegee-tokyo, 2024-12-05:
LMH_SUCCESS only means that the packet has been successfully enqueued for transmission. Nothing else.

If you want to know whether the packet was sent and in case of confirmed packets, you need to use the above callbacks.

`bool result` is true when packet was confirmed by LNS with an ACK. Otherwise it is false.

JDH023, 2024-12-05:
Hello.

It is working as expected, thanks.

closed 2024-12-05 by JDH023