beegee-tokyo/SX126x-Arduino #139: OTAA with ISP4520

ericlangel, 2025-02-18:
Hello,

i have an ISP4520 Module and a Dragino LPS8 Gateway
APB is working
but with OTAA i cant join.
App Key and Device EUI are both MSB

here is my hwconfig:
```
 // Define the HW configuration between MCU and SX126x
  hwConfig.CHIP_TYPE = SX1261_CHIP;     // Example uses an eByte E22 module with an SX1262
  hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
  hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;   // LORA SPI CS
  hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // LORA SPI CLK
  hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // LORA SPI MISO
  hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
  hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // LORA SPI BUSY
  hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // LORA SPI MOSI
  hwConfig.RADIO_TXEN = RADIO_TXEN;     // LORA ANTENNA TX ENABLE
  hwConfig.RADIO_RXEN = RADIO_RXEN;     // LORA ANTENNA RX ENABLE
  hwConfig.USE_DIO2_ANT_SWITCH = true;    // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
  hwConfig.USE_DIO3_TCXO = false;        // Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
  hwConfig.USE_DIO3_ANT_SWITCH = true;   // Only Insight ISP4520 module uses DIO3 as antenna control
  hwConfig.USE_RXEN_ANT_PWR = false;
```

are there any known issues with ISP4520?

beegee-tokyo, 2025-02-18:
If you can connect with ABP, then your connection is correct.
My guess is
- is the device set to OTAA mode?
- is the device registered as OTAA device in your LoRaWan server?
- did you make sure that the DevEUI, AppEUI and AppKey of the device are the same used in your LoRaWAN server when you registered the device there?

ericlangel, 2025-02-18:
> is the device set to OTAA mode?
yes
`err_code = lmh_init(&lora_callbacks, lora_param_init, true, CLASS_A, LORAMAC_REGION_EU868);`

> is the device registered as OTAA device in your LoRaWan server?
yes, we have a lot (more than 1000) of LoRaWan devices with ABP and OTAA. Its our first device with ISP4520 and your Lib

> did you make sure that the DevEUI, AppEUI and AppKey of the device are the same used in your LoRaWAN server when you 
> registered the device there?
i got the DevEUI and AppKey from our Server 
i have no AppEUI. On a Nrf52840 with RFM95 an other Library i didnt needed it.

here the Debug Output:
10:23:46.672 -> =====================================
10:23:46.672 -> SX126x LoRaWan test
10:23:46.672 -> =====================================
10:23:46.672 -> <BRD> SyncWord = 2414
10:23:46.672 -> lora_hardware_init OK- 0
10:23:46.672 -> <LMH> <BRD> OTAA 
10:23:46.672 -> DevEui=36-F7-93-53-20-83-F3-16
10:23:46.672 -> DevAdd=014364D8
10:23:46.672 -> AppEui=00-0LoRa Task started0-00-00-00-00-00-00
10:23:46.672 -> AppKey=AB-D0-9B-70-F2-D2-D5-11-16-4B-30-0F-2
10:23:46.672 -> C-55-83-5E
10:23:46.764 -> <LMH> Selected subband 1
10:23:46.764 -> lmh_init OK - 0
10:23:46.764 -> <LMH> Selected subband 2
10:23:46.764 -> <EU868> Channel count ch# 8, freq 0
10:23:46.764 -> <EU868> Channel count ch# 9, freq 0
10:23:46.764 -> <EU868> Channel count ch# 10, freq 0
10:23:46.764 -> <EU868> Channel count ch# 11, freq 0
10:23:46.764 -> <EU868> Channel count ch# 12, freq 0
10:23:46.764 -> <EU868> Channel count ch# 13, freq 0
10:23:46.764 -> <EU868> Channel count ch# 14, freq 0
10:23:46.764 -> <EU868> Channel count ch# 15, freq 0
10:23:46.764 -> <EU868> Channel count ch# 0, freq 868100000
10:23:46.811 -> <EU868> Set channel 0, frequency 868100000
10:23:46.811 -> <EU868> Channel count ch# 1, freq 868300000
10:23:46.811 -> <EU868> Set channel 1, frequency 868300000
10:23:46.811 -> <EU868> Channel count ch# 2, freq 868500000
10:23:46.811 -> <EU868> Set channel 2, frequency 868500000
10:23:46.811 -> <EU868> Channel count ch# 8, freq 0
10:23:46.811 -> <EU868> Channel count ch# 9, freq 0
10:23:46.811 -> <EU868> Channel count ch# 10, freq 0
10:23:46.811 -> <EU868> Channel count ch# 11, freq 0
10:23:46.811 -> <EU868> Channel count ch# 12, freq 0
10:23:46.811 -> <EU868> Channel count ch# 13, freq 0
10:23:46.811 -> <EU868> Channel count ch# 14, freq 0
10:23:46.811 -> <EU868> Channel count ch# 15, freq 0
10:23:46.857 -> <EU868> Using channel 0, frequency 868100000
10:23:46.857 -> <LM> dutyCycleTimeOff was = 0
10:23:46.857 -> <LM> Set send_join_now to false
10:23:49.848 -> <RADIO> TimerTxTimeout
10:23:49.848 -> <LM> OnRadioTxTimeout
10:23:49.848 -> <LM> LoRaMacState = idle
10:23:49.848 -> OVER_THE_AIR_ACTIVATION failed!
10:23:49.848 -> Check your EUI's and Keys's!
10:23:49.848 -> Check if a Gateway is in range!

ericlangel, 2025-02-19:
any idea?

beegee-tokyo, 2025-02-20:
Is the device registered as OTAA device in your LoRaWAN server?

ericlangel, 2025-02-21:
yes.

We also tried a working DevEUI and AppKey of a different device which was working before.
With this 2nd key i also tried all 4 combinations of LSB/MSB

beegee-tokyo, 2025-02-22:
I don't know.
I am using this library every day with nRF52840 devices without problems.
I do not have anymore ISP4520 modules to test

fabltd, 2025-03-07:
Did you fix this.

It seems that the Lib has stopped connecting.

beegee-tokyo, 2025-03-08:
Try an older version.
It works fine here in my everyday projects with nRF52840, ESP32 and RP2040 MCU's.
I can't test nRF52832, I don't have anymore boards with it.

closed 2025-05-14 by beegee-tokyo