beegee-tokyo/SX126x-Arduino #143: Persisting LoRaWAN state across device restarts - how to access CFList?

netmaniac, 2025-03-10:
Hello
I'm trying to persist LoRaWAN state to not require new OTAA  join request after device reboot (no battery power). Using hints from #85 I managed to store all mentioned data (network and app session keys, uplink and downlink counter) and it is almost working. 

Almost, since I don't know how to store CFlist data and later restore. Result? After reboot ADR is not working, device uses only 3 basic frequencies (EU 868 region, so uplinks after reboot are being sent only on 868.1, 868.3 and 868.5). Also downlinks seems not reaching device.

When device joins with OTAA, until first reboot everything is fine - ADR is working (it will be stationary device), node sends uplinks using all 8 frequencies and downlinks are delivered to node.

How to persist CFList and then re-apply?

beegee-tokyo, 2025-03-11:
The additional channels are added in [RegionEU868ApplyCFList()](https://github.com/beegee-tokyo/SX126x-Arduino/blob/9e6717ffe60a282985745578f7934126f56d414e/src/mac/region/RegionEU868.cpp#L425).
That could be a starting point. But the library does not support what you want to do, you will have to do some patches.

For the other problems, I don't know. What you are doing is not a standard usage.

netmaniac, 2025-03-11:
OK, I'll try then get those data and have node which can be safely turned off or put in deep sleep and will return with results.

closed 2025-03-11 by netmaniac