beegee-tokyo/SX126x-Arduino #149: Incompatibility with AI Thinker Ra-01SH (SX1262) — Ping-Pong Example Fails to Send/Recv Messages

NaufalDhaffa, 2025-05-14:
I am encountering an issue when using your library with the AI Thinker Ra-01SH module, which is based on the Semtech SX1262 LoRa chip. The problem arises specifically when running the ping-pong communication example.
The module fails to send any messages in the ping-pong example. The sender seems to get stuck while trying to transmit, and no response is received from the other node. 
[image]
Module: AI Thinker Ra-01SH (SX1262)
Microcontroller: ESP32
connection :
LoRa RA-01SH:
  GND  -> GND
  3V3  -> 3.3V
  GPIO5  -> NSS
  GPIO14 -> RST 
  GPIO33 -> BUSY 
  GPIO18 -> SCK
  GPIO19 -> MISO
  GPIO23 -> MOSI
  GPIO25 -> DIO1 
  GPIO26 -> TXEN 
  GPIO27 -> RXEN

beegee-tokyo, 2025-05-14:
Please check the update example for LoRa P2P [PingPong](https://github.com/beegee-tokyo/SX126x-Arduino/tree/master/examples/PingPong) and [PingPongPio](https://github.com/beegee-tokyo/SX126x-Arduino/tree/master/examples/PingPongPio)

The original example seems to have problems with ESP32.
But I could test the new code only with a ESP32 Wrover (RAK11200+RAK13300) from RAKwireless. I don't have other ESP32 modules for testing.

For the module and it's connections, I don't know the AI-thinker board.

NaufalDhaffa, 2025-05-14:
> Please check the update example for LoRa P2P [PingPong](https://github.com/beegee-tokyo/SX126x-Arduino/tree/master/examples/PingPong) and [...]

what have you changed from the code?

NaufalDhaffa, 2025-05-14:
it still didnt work

beegee-tokyo, 2025-05-15:
Can't help further.
I tried the new PingPong on an ESP32 Wrover and a ESP32-S3 and it worked,

I changed the whole PingPong example to use Semaphores and handle the events in the loop instead of handling directly in the callbacks.

I am guessing you have a hardware or connection issue.

closed 2025-05-16 by beegee-tokyo