beegee-tokyo/SX126x-Arduino #153: ESP32 with Wio SX1262 configuration

JuvanBroos, 2025-07-23:
I am trying to get the PingPong example working on two PCBs that I designed myself that utilizes a ESP32C6 chip and the Wio SX1262 module. I get the following result however:

\=====================================
SX126x PingPong test
\=====================================
MCU Espressif ESP32
BoardId: FC-D6-06-FE-FF-04-32-54
Starting lora_hardware_init
LoRa init success
Starting Radio.Rx
OnRxTimeout CB
Loop wakeup
OnRxError
Send as Master after RX timeout

and nothing happens afterwards. Here is my configuration settings that I have altered from the example: (The Wio SX1262 RF_SW is connected to pin21 on my board and the rest is as shown in the code)
```
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 20;	 // LORA RESET
int PIN_LORA_DIO_1 = 23; // LORA DIO_1
int PIN_LORA_BUSY = 22;	 // LORA SPI BUSY
int PIN_LORA_NSS = 18;	 // LORA SPI CS
int PIN_LORA_SCLK = 6;	 // LORA SPI CLK
int PIN_LORA_MOSI = 7;	 // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 21;	 // LORA ANTENNA RX ENABLE

#define RF_FREQUENCY 868000000	// Hz

hwConfig.CHIP_TYPE = SX1262_CHIP;		  // Example uses an eByte E22 module with an SX1262
hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	  // LORA SPI CS
hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;	  // LORA SPI CLK
hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;	  // LORA SPI MISO
hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;	  // LORA SPI BUSY
hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;	  // LORA SPI MOSI
hwConfig.RADIO_TXEN = RADIO_TXEN;		  // LORA ANTENNA TX ENABLE
hwConfig.RADIO_RXEN = RADIO_RXEN;		  // LORA ANTENNA RX ENABLE
hwConfig.USE_DIO2_ANT_SWITCH = true;	  // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
hwConfig.USE_DIO3_TCXO = true;		// Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
hwConfig.USE_DIO3_ANT_SWITCH = false;	  // Only Insight ISP4520 module uses DIO3 as antenna control
hwConfig.USE_RXEN_ANT_PWR = false;		  // RXEN is used as power for antenna switch
```

The confusing bit is on the Wio SX1262 datasheet regarding what is RXEN and TXEN and whether RXEN is used as power for the RF Switch. I have attached some screenshots:

[image]

## Pin Definitions
[image]

## Electrical Characteristics
[image]

Where would the problem be? Are my configuration settings wrong? The SPI seems to be working. It seems as if it never exists CAD mode.

beegee-tokyo, 2025-07-23:
I can't say anything about the Wio Module, I don't know it.

Did you try 
```
hwConfig.USE_RXEN_ANT_PWR = true;		  // RXEN is used as power for antenna switch
```
And check DIO1 and BUSY assignment. DIO1 is used as interrupt input on the MCU. It is triggered from the SX1262 after an event (RX/TX/timeout/CRC error, ...

I don't have an ESP32C6, not sure if there are any other differences to the ESP32-Wrover or ESP32-S3. I tested with this two versions and it works.

JuvanBroos, 2025-07-23:
I have tried that without success and checked the Busy and DIO1 assignment. What does it mean when it says OnRxError? What is supposed to happen after Radio.StartCad(); ? How long is the CAD process supposed to take?

JuvanBroos, 2025-11-14:
I have found my issue. One was that the ESP32C6 was not soldered on properly and some connections were not working. The other was that the library was trying to use the HSPI bus by default and that was causing the SPI communication not to work. So I changed the spi_board.cpp file for the espressif mcu on my side to the following:

```cpp
#if defined ESP8266 || defined ESP32
#include <SPI.h>
#include "boards/mcu/board.h"

SPIClass SPI_LORA(FSPI); // changed to FSPI for the ESP32C6 to work

void initSPI(void)
{
#ifdef ESP8266
	SPI_LORA.pins(_hwConfig.PIN_LORA_SCLK, _hwConfig.PIN_LORA_MISO, _hwConfig.PIN_LORA_MOSI, _hwConfig.PIN_LORA_NSS);
	SPI_LORA.begin();
	SPI_LORA.setHwCs(false);
#else
	SPI_LORA.begin(_hwConfig.PIN_LORA_SCLK, _hwConfig.PIN_LORA_MISO, _hwConfig.PIN_LORA_MOSI, _hwConfig.PIN_LORA_NSS);
#endif
}
#endif
```